lpc-rs/lpc8xx-hal #14: Make runtime support more convenient

hannobraun, 2017-11-28:
Currently, the user must specify the MCUs memory layout (by adding `memory.x`). This could be made more convenient, by allowing the user to just specify which part they use (maybe using a Cargo feature?) and somehow creating the memory layout for [cortex-m-rt](https://crates.io/crates/cortex-m-rt) automatically.

hannobraun, 2020-01-08:
There's some work on this in #187, although that selects a conservative configuration that should fit all target devices. It would be better, if it selected a correct configuration based on Cargo features specified.

david-sawatzke, 2020-01-08:
I'm not sure if that's a good idea to select a memory.x file based on features, especially since we'd need to add an additional feature differentiating between lpc822 & lpc824.

I also think not "hiding" the need for a memory.x file is the better choice for hals, but that's just personal preference.

hannobraun, 2020-01-09:
> I'm not sure if that's a good idea to select a memory.x file based on features, especially since we'd need to add an additional feature differentiating between lpc822 & lpc824.

I think having additional features to differentiate between hardware differences is a good thing. That way, the code can be configured to provide exactly what the user needs, and prevent accidental use of features that aren't actually there.

Granted, I haven't looked at the differences between LPC822 and LPC824 lately, so I don't need how relevant this would be besides memory. But I think memory is a good enough reason to add those features, and I don't think they would be too much of a burden (existing hardware doesn't change much, after all).

> I also think not "hiding" the need for a memory.x file is the better choice for hals, but that's just personal preference.

Two notes:
- Right now, a first point of contact between new users and LPC8xx HAL is an obscure error message about `memory.x`. We can fix that problem, and I think we should.
- For most users (i.e. those running a single application that doesn't do anything weird memory-wise), there really is no need for providing their own `memory.x`, if the correct one is selected for their hardware. Forcing users to figure this out themselves is neither necessary nor user-friendly.

Would your concerns be alleviated, if I extended my pull request to provide the correct configuration for all supported targets?

> I think we should first let the discussion play out in matrix.

I don't like this suggestion :-)

First, I don't have the time nor inclination to follow that discussion. I think the format is good for socializing and real-time stuff, but not for drawn-out technical discussions.

Second, as far as I did follow it, the discussion was about shortcomings in STM32L0 HAL. We can simply do a better job here.

If some ecosystem-wide consensus emerges later, we can adapt, but for now, there's a real problem in this crate (everyone having to provide the same configuration again and again) that we can fix in this crate.

hannobraun, 2020-01-09:
Update: The number of ADC channels and GPIO pins differs significantly within the 82x family. I think it's definitely worth reflecting that in the HAL, to prevent accidental use of non-existing features.

david-sawatzke, 2020-01-09:
> First, I don't have the time nor inclination to follow that discussion.

That's quite understandable

It seemed to end in an agreement when using the hal provided memory.x, but no implementation of that yet.

> If some ecosystem-wide consensus emerges later, we can adapt

Seems good to me

> Would your concerns be alleviated, if I extended my pull request to provide the correct configuration for all supported targets?

It's not that important to me, I'm fine with merging it as is and if I/the larger ecosystem comes with something better, we can still change it (since we're still far off from a 1.0)

david-sawatzke, 2020-01-09:
> Update: The number of ADC channels and GPIO pins differs significantly within the 82x family. I think it's definitely worth reflecting that in the HAL, to prevent accidental use of non-existing features.

That's also the case with the lpc845 variants. I think the approach of ignoring gpio pin differences is the right one, but the lpc845 variants also differs in peripherals.

This is currently not an issue, but if we implement DAC or CAPT, we need to add more features to differentiate between them.

hannobraun, 2020-01-09:
I updated #14. It probably makes sense to continue the discussion there.

>  I think the approach of ignoring gpio pin differences is the right one

I'm not so sure myself, but we can discuss when we get to that :-)

hannobraun, 2020-01-10:
This was addressed in #187.

closed 2020-01-10 by hannobraun