lpc-rs/lpc8xx-hal #53: Pin output API might be too restrictive

hannobraun, 2018-02-23:
The current API only allows for setting the output level when the pin is in the output state. This seems sensible at first, but it might cause the wrong output level to be set for a moment, right after the pin has entered the output state, but right before the desired output level has been set. Depending on the application, this might be a problem.

I think it may be better to make `set_high` and `set_low` available regardless of pin direction. However, if that change is made, I wonder if it still makes sense to encode the pin direction as a type parameter at all.

I'd love to hear more opinions about this design issue! Please leave a comment, especially if you have relevant practical experience.

somemetricprefix, 2018-02-23:
Without knowing the current design:
Maybe pass an initial state to the function that configures the PIN as output.
HAL then updates the state before changing to output mode.
This inital state could be high, low or unspecified. unspecified would then just skip to change the state before configuring the pin as output.

hannobraun, 2018-02-23:
@somemetricprefix Yeah, that could work. If we use a generic parameter for the initial state, we could even make it zero-cost, if "unspecified" is passed. Thank you!

commit 08631450 (2020-03-12): Set initial level when switching to output mode

Previously, there was no way to control the initial output level of a
pin, when switching it to output mode. This doesn't matter for a
blinking LED, but there are situations where an undesired signal is not
acceptable.

This commit rectifies this, by requiring the initial level to be passed
as an argument. This is a bit more tedious for the common case, but more
flexible, and overall less error-prone (I've had bugs in the past, that
were caused by me not thinking of the initial output level).

Close #53
closed 2020-03-12 by hannobraun