lpc-rs/lpc8xx-hal #137: LPC845 support

david-sawatzke, 2019-03-06:
The lpc845 has a pretty nifty little [dev board](https://www.arrow.com/en/products/lpc845-brk/nxp-semiconductors), and it seems pretty similar to the lpc82x. The remaining peripheral differences seem pretty easy to feature-gate, but unfortunately lots of fields are differently named in the `lpc845.svd` compared to the old `lpc82x.svd`, so it's not that easy. 

Fortunately the new `lpc82x.svd` seems more similar, even though there are still some differences left.

A fork seems like a bad idea and I think if we use the new `lpc82x.svd` file, the feature gates aren't going to be too bad. Since the `lpc845.svd` file has to be fixed anyway (since i there doesn't seem to exist an old-variant file) and the new `lpc82x.svd` file is pretty similar, that's seems like the way to go.

hannobraun, 2019-03-07:
I'd love to extend the `lpx82x-hal` code base to support more devices from the LPC800 family!

Before I address your comments specifically, I'd like to lay out a plan for how that could be done. I think the main problem in supporting a cross-device HAL is re-using as much code as possible, without making the development or usage for any specific device too complicated. The best approach that I know to achieve that is the one used in [`nrf52-hal`](https://github.com/nrf-rs/nrf52-hal). Basically, it works like this:

- There is one crate for common code. In our case this would be called `lpc8xx-hal-common`, or `lpc800-common`. The purpose of this crate is to be a building block for more specific crates, not to be used by end users.
- There are additional crates for each device: `lpc82x-hal` and `lpc84x-hal`. Those depend on `lpc8xx-hal-common` and re-export common code (via `pub use`) in their own API.

I think this approach is very attractive because it makes it easy to share code, while at the same code making it very easy to write device-specific code, where that's necessary. For example, if someone contributes a peripheral API, but can only test on LPC84x, it can be submitted to `lpc84x-hal`, and ported to `lpc8xx-hal-common` later.

Let me know what you think of this approach. I think we have made good experiences with it over at [`nrf52-hal`](https://github.com/nrf-rs/nrf52-hal). The main problem is probably that there's some duplication between the crates (Cargo.toml and such), and that there are multiple crates to build and publish.

---

> A fork seems like a bad idea and I think if we use the new `lpc82x.svd` file, the feature gates aren't going to be too bad. Since the `lpc845.svd` file has to be fixed anyway (since i there doesn't [...]

I've attempted to use the new SVD file before, and it was a bit of a disaster. It has a large number of subtle problems, and I decided it wasn't worth to pursue further ([details](https://community.nxp.com/thread/484757)).

That said, if it's going to help with supporting LPC84x, maybe it's worth giving it another try. The file would need pretty much a full review though. I had hoped that NXP would do that for us, but I don't think it's wise to rely on that :-)

david-sawatzke, 2019-03-07:
I've mainly used the stm32f0 hal, which just used features, but the peripherals, register & field names were more similar, so that's probably not going to work that well for the lpc8xx series.

I concur that the nrf52-hal way sounds like the best way to approach this, especially since there do exist some differences.

I've checked the lpc845 svd file to the old lpc82x one via ediff and the missing interrupts seemed like the only dealbreaker. A lot of the reset masks were wrong on the old file as well (at least for the lpc845). 

I think going over both files with some kind of diff tool that can handle the differences is probably the best approach to find flaws and it "only" took a few hours.

hannobraun, 2019-03-08:
Okay, so it seems we're in agreement about the general approach.

Here's what I think the next steps need to be:

- [ ] Move `lpc82x-hal` into a `lpc82x-hal` subdirectory, create `lpc8xx-hal-common`, make `lpc82x-hal` depend on it. For a start, the `lpc8xx-hal-common` crate could be left empty.
- [ ] Figure out stuff like global README, global build script, update Travis config, etc.
- [ ] Add `lpc845-hal`. Add it to README, build script, etc.

It might be best to group the previous items into one pull request.

- [ ] Rename the repository to `lpc8xx-hal`, make any configuration changes required.
- [ ] Start moving stuff from `lpc82x-hal` to `lpc8xx-hal-common` and re-use it in `lpc845-hal`. I expect this will be a longer process, spread over multiple pull requests.

I won't have time to do any of this myself in the short term, but I'm happy to assist with feedback, reviews, etc.

hannobraun, 2019-04-11:
I'm closing this issue, as LPC845 support now exists. Further improvements are extremely welcome, of course, but I don't think we need this issue to track them.

Thank you again for all your work, @david-sawatzke!

closed 2019-04-11 by hannobraun