lpc-rs/lpc8xx-hal #159: Compatibility of pin_state::Gpio with RTFM?

nickray, 2019-08-31:
Might be my RTFM-noob-factor in play, but: Are the references taken in `pin_state::Gpio` intended to be used with RTFM? I can't seem to get the [example](https://github.com/nickray/lpc55s6x-hal/blob/ebbb8cf1d41c508ae983459560f3be63bc9adbdf/examples/rtfm_led.rs#L17) I'm working on to compile.

hannobraun, 2019-09-01:
I don't currently have time to look into this issue in detail (currently on vacation, will be back in the office September 9). But I can say the following:
- Nothing in this HAL is intended to be used with RTFM. I did most of the design before I was familiar with RTFM.
- There's stuff that definitely doesn't work with RTFM (see https://github.com/japaric/cortex-m-rtfm/issues/163 for some context).
- I don't know of any examples combing this HAL with RTFM. I'd be surprised if they existed, because I don't think it's possible right now.
- I haven't had a chance to do any significant work on this HAL for a while now, and the design certainly doesn't reflect my current thinking.
- Most relevantly to your question, one thing I'd change is that I'd go to great lengths to avoid any long-lived references. They mess with too many use cases.
- I intend to do a big redesign, but that might be a while off.

Hope that helps answer your question. I'd be happy to help you out further, so feel free to open more issues. Just don't expect timely or detailed replies before September 9.

nickray, 2019-09-04:
Thanks. I went ahead and made some changes (e.g. a `RegClusterProxy`). First version is published: https://crates.io/crates/lpc55s6x-hal. I added you as co-author, hope you don't mind :)

closed 2019-09-04 by nickray

hannobraun, 2019-09-11:
I have to look at your work more closely at some point!

> I added you as co-author, hope you don't mind :)

Thanks, that's fine :-)

nickray, 2019-09-11:
Thanks for your interest, not sure it makes too much sense for you to look closely currently though, as the design is in a state of semiconstant flux currently. As a small example, not sure I agree Pins should be part of a split SWM/IOCON, currently I think they could be separate (with private state variable tracking if they're "taken"); similar for other objects that materialize as side effect of the HAL wrapping a raw peripheral.

Will certainly ask you (directly) for review once I have something I like!