lpc-rs/lpc8xx-hal #182: Deprecate lpc82x-hal

david-sawatzke, 2020-01-03:
@hannobraun There's no notice on [crates.io](https://crates.io/crates/lpc82x-hal) that it's deprecated

hannobraun, 2020-01-06:
Good catch. I'll take care of it.

david-sawatzke, 2020-01-08:
Something minor thing: The github blurb still suggests that this crate is only for lpc82x

hannobraun, 2020-01-09:
> Some minor thing: The github blurb still suggests that this crate is only for lpc82x

Fixed.

david-sawatzke, 2020-06-19:
Had a quick look over the issues, ping @hannobraun

hannobraun, 2020-06-19:
Thanks, I haven't forgotten. I've just been busy enough these past months that everything not on the critical path has fallen by the wayside.

Looks like things might start to clear up for me soon, so I hope this won't stay open much longer.

hannobraun, 2020-08-14:
Done!

closed 2020-08-14 by hannobraun