lpc-rs/lpc8xx-hal #183: Investigate SWM weirdness

david-sawatzke, 2020-01-06:
Even though, according to the reference manual, swm should be enabled per default, it needs to be reenabled in the [usart example](https://github.com/lpc-rs/lpc8xx-hal/blob/8ca1a9ce19eef65904695a24bb6113420578966d/examples/usart.rs#L20) for the lpc845.

It shouldn't be possible to use it in a disabled state

david-sawatzke, 2020-01-06:
It's also not enabled per default on hardware:
```
(gdb) x  0x40048080
0x40048080:	0x00000017
```
So it seems the RM really is wrong

hannobraun, 2020-01-07:
Sounds like there are two action items here:
- Fix our API
- Report the bug to NXP

david-sawatzke, 2020-01-13:
The issue was closed, since it's fixed for the hal, but we should still report it

hannobraun, 2020-01-13:
Yes, good point. Updated title and labels accordingly.

commit 09f57525 (2020-01-13): Merge pull request #190 from david-sawatzke/swm-enabled

Force swm to be enabled on lpc845, fix #183
commit ab17d311 (2020-01-13): Force swm to be enabled on lpc845, fix #183