lpc-rs/lpc8xx-hal #211: Consider getting rid of `RegProxy`

hannobraun, 2020-02-18:
It was added to work around some perceived shortcomings in svd2rust (see https://github.com/rust-embedded/svd2rust/issues/213), and while I still think it would be worthwhile to make the proposed improvements in svd2rust, `RegProxy` causes a lot of complexity in this crate, and its utility is dubious at best.

hannobraun, 2020-02-18:
Some more thoughts: https://github.com/lpc-rs/lpc8xx-hal/pull/210#issuecomment-587406534

david-sawatzke, 2020-02-18:
I think in it's current incarnation (ignoring the safety issues), `RegProxy` is of limited use. In `syscon.rs` it makes what's happening more clear, whilst in #210 it makes the code more complex & confusing. 

Getting rid of  `RegProxy` & replacing it with the `unsafe` equivalent, when it's used for shared registers (in ctimer, #210 and parts of dma) makes sense, but for the syscon it's seems fine if we just declare the constructor as `unsafe`.

hannobraun, 2020-02-19:
Sounds like a good compromise actually. I've added a task to my list to do just that (at the current pace, I think it'll be a few weeks).

It would be great to make `RegProxy` completely redundant by improving svd2rust, but there's so much to do... maybe some day.