lpc-rs/lpc8xx-hal #304: building examples yields I/O error on master

Lotterleben, 2021-01-08:
When I call

```console
lpc8xx-hal git:(master) cargo embed lpc845 --example gpio_delay --features 845-rt
```

on the current master, I get the following error message:

```console
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
    Flashing /Users/lottesteenbrink/client projects/GF/lpc8xx-hal/target/debug/examples/gpio_delay
     Erasing sectors ⠁ [00:00:00] [#########################################]       0B/      0B @       0B/s (eta 0s )
       Error failed to flash /Users/lottesteenbrink/client projects/GF/lpc8xx-hal/target/debug/examples/gpio_delay

Caused by:
    0: I/O error
    1: No such file or directory (os error 2)
```

It seems that https://github.com/lpc-rs/lpc8xx-hal/commit/db334f6b04b05721a3e2531b0141d3229038de03 somehow introduced this– when I rename the file back to `.cargo/config` (without the .toml), it works on `master` too. Not quite sure what's going on there...

david-sawatzke, 2021-01-08:
Maybe an old `cargo-embed` version? See https://github.com/probe-rs/cargo-embed/issues/29. This works for me™.

hannobraun, 2021-01-09:
It worked for me too (unless I accidentally tested the wrong branch).

Lotterleben, 2021-01-12:
ah yep, that was it! I'm still getting accustomed to things moving faster in Rust land, my apologies.

closed 2021-01-12 by Lotterleben