lpc-rs/lpc8xx-hal #307: usart rx: OVERRUN flag is never cleared

Lotterleben, 2021-02-22:
The`OVERRUN` flag is never cleared in https://github.com/lpc-rs/lpc8xx-hal/blob/master/src/usart/rx.rs#L274, making it difficult to recover from a USART overrun.
We've found that modifying `Rx::receive` to clear the flag resolves the issue:
```diff
if stat.overrunint().bit_is_set() {
+    usart.stat.write(|w| w.overrunint().set_bit());
    Err(nb::Error::Other(Error::Overrun))
 }
```

however, there's also the (unused in the hal) function `is_flag_set()` that according to its docs (and from what I understand about its implementation) clears the flag that's being checked as a side effect. Unfortunately, I've been unable to confirm that 

```diff
-if stat.overrunint().bit_is_set() {
+if self.is_flag_set(Flag::OVERRUN) {
    Err(nb::Error::Other(Error::Overrun))
 }
```

resolves the issue as well, but tbh my testing strategies were a bit wonky.

hannobraun, 2021-02-23:
I haven't checked against the user manual, but this sounds legit. We probably should check the other flags too, as probably more of them need to be reset manually.

But I don't think `is_flag_set` is going to work. As the comment above the line you linked states, the register needs to be read all at once. `is_flag_set` seems to do one register read per flag (which might not be the best API design).

Lotterleben, 2021-02-23:
Ack to all of this :) The Overrun flag is currently the only one that's blocking us, though– can we find a way to fix this that we're happy with and then look into and adjust the others in the same manner later?

david-sawatzke, 2021-02-23:
I've looked at it and it seems like the comment is only referring to the `RXDATSTAT` register, which clears `RXRDY` on read

```
if /* usart overrun*/ {
    usart.rx.is_flag_set(flags.Flag.OVERRUN)
}
```
should™ work just fine in user code.

Not sure if we want to automatically clear it when returning an error, but exposing something better is probably a good idea.

hannobraun, 2021-02-24:
@Lotterleben 
> Ack to all of this :) The Overrun flag is currently the only one that's blocking us, though– can we find a way to fix this that we're happy with and then look into and adjust the others in the same manner later?

Definitely. I'll merge anything that is an improvement, even if it doesn't address all potential problems.

@david-sawatzke 
> I've looked at it and it seems like the comment is only referring to the `RXDATSTAT` register, which clears `RXRDY` on read

Ah, right, I agree.

Still, calling `self.is_flag_set` (without further improvement to that API) would read STAT again. Not sure if that's the right thing to do.

> Not sure if we want to automatically clear it when returning an error, but at least exposing something better is probably a good idea.

Clearing the error flag when returning an error to the user seems right to me. I think that's most convenient (and least surprising) to the user, and I can't think of a reason not to.

> EDIT: What should we do with the data in the fifo? Discard it on an overrun, so there's a clean break between pre-overrun & post-overrun?

Discarding the data would probably work for most cases, but I'm not sure it's the optimal behavior. Maybe return it with the error? That provides the most flexibility to the user.

hannobraun, 2021-02-25:
I've taken a stab at the problem: https://github.com/lpc-rs/lpc8xx-hal/pull/308

Let me know what you think.

PR #308 (2021-02-26): Improve USART error handling
I've verified that receiving characters via USART still works with this code. But I don't have a setup that would allow me to inject errors, so I can't test whether this has the intended effect. All feedback is appreciated!

Close #307
closed 2021-02-26 by hannobraun