lpc-rs/lpc8xx-hal #334: Problems attaching debugger

mxdamien, 2022-01-15:
Hi !

I've been trying this the last few days and I'd like to apologize in advance in case this is a stupid question.

I have a LPC845-BRK Devboard and want to try out Rust in an embedded environment. I am able to flash one of the samples and also my own small toy program onto the board with cargo flash ( e.g. cargo flash --chip=LPC845M301JBD48 --features=845-rt --example=gpio_timer ) and see it working. But I struggle to attach a debugger. 

When running 'cargo run' I get this output from gdb:

user@pc:~/arm/lpc8xx-hal$ cargo run --example=gpio_timer --features=845-rt
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `arm-none-eabi-gdb -q -x openocd.gdb target/thumbv6m-none-eabi/debug/examples/gpio_timer`
Reading symbols from target/thumbv6m-none-eabi/debug/examples/gpio_timer...
Open On-Chip Debugger 0.11.0+dev-00551-gaad871805 (2022-01-14-10:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
openocd.gdb:1: Error in sourced command file:
Remote communication error.  Target disconnected.: Connection reset by peer.

and this message in openocd.log:

Error: unable to find a matching CMSIS-DAP device

I also tried 'cargo embed':

user@pc:~/arm/lpc8xx-hal$ cargo embed --example=gpio_timer --features=845-rt
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
      Config default
      Target /home/david/arm/lpc8xx-hal/target/thumbv6m-none-eabi/debug/examples/gpio_timer
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0x5000c000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0x5000c000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0x5000c000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
       Error failed attaching to target
             
Am I doing something wrong or am I missing something?

Regards 
David

hannobraun, 2022-01-15:
Hey @mxdamien! First off, you shouldn't have to run `cargo flash` before `cargo run`. OpenOCD should flash the program (if it worked), so a separate `cargo flash` is redundant.

Second, that OpenOCD configuration is really old, and I'm not sure if it still works. It might be better to just remove it from the repository. OpenOCD was always a huge source of problems for me, and I switched to probe-rs when it became practical to do so.

I recommend that any modern Embedded Rust setup use probe-rs, either in the form of `cargo embed` (as configured in this repository) or `probe-run`.

Third, I see two things that are missing with your `cargo embed` setup:
1. Our [Embed.toml](https://github.com/lpc-rs/lpc8xx-hal/blob/master/Embed.toml) specifies two profiles, for 82x and 845 respectively, and you need to select one (see README). Try `cargo embed lpc845 --example=gpio_timer --features=845-rt`.
2. To attach a debugger to cargo-embed, you need to enable that capability in the configuration, which our setup doesn't do. Try adding [something like this](https://github.com/probe-rs/cargo-embed/blob/6a3ac6b033098e005169c2af6f58243db6d8cb43/src/config/default.toml#L67-L71) to the configuration (but with `enabled = true`, of course).

Hope that helps! If you still have questions, feel free to ask here, but actually the [probe-rs Matrix channel](https://matrix.to/#/#probe-rs:matrix.org) might be a better place, as I've never used GDB and cargo-embed together.

mxdamien, 2022-01-15:
Hi Hanno,

thank you very very much for the help. I managed to get it working. After your hint I had a look at robe-rs and ditched the idea of using OpenOCD / GDB in favour of it. After a few minutes I did in fact see my first breakpoint being hit in VSCode .

thanks again -- Have a nice weekend.

David

closed 2022-01-15 by mxdamien